chris1111/Wireless-USB-Adapter #1: Possible Support for EAP-TTLS Authentication using WPA2 Enterprise?

Steve-luo, 2018-11-11:
My school wifi uses WPA2 Enterprise with EAP-TTLS authentication. Only a username and a password are used, no certificate required. Is it possible for this driver to support that?

chris1111, 2018-11-11:
> My school wifi uses WPA2 Enterprise with EAP-TTLS authentication. Only a username and a password are used, no certificate required. Is it possible for this driver to support that?

I don't know? you have to try and let me know if that work

Steve-luo, 2018-11-11:
> > My school wifi uses WPA2 Enterprise with EAP-TTLS authentication. Only a username and a password are used, no certificate required. Is it possible for this driver to support that?
> 
> I don't know? you have to try and let me know if that work

Oh I should clarify, It doesn't support EAP-TTLS. The only two authentication options are PEAP and EAP-TLS, and neither of them work. So I'm asking if it's possible to add support for it in future updates. (And thanks for the quick reply! )

chris1111, 2018-11-11:
> Oh I should clarify, It doesn't support EAP-TTLS. The only two authentication options are PEAP and EAP-TLS, and neither of them work. So I'm asking if it's possible to add support for it in future [...]

Sorry
I do not think that for this drivers it will be possible

Steve-luo, 2018-11-11:
> Sorry
> I do not think that for this drivers it will be possible

Yeah, I gathered it's more of a software thing instead of the driver. But still, I really appreciate your efforts in making this and updating.

closed 2018-11-11 by Steve-luo

MuntashirAkon, 2019-02-13:
@Steve-luo The underlying framework behind this app is actually a famous tool called **[wpa_supplicant](https://w1.fi/wpa_supplicant/)** and is actively being used in Linux-based distros and Android devices. I don't know (have to investigate) but it may be compiled with EAP-TTLS support. You can search the internet in order to add necessary configurations to `/etc/wpa_supplicant.conf` for this to work (I cannot guarantee, though).

**Binary file location** (if you've already installed this software): `/Library/Application\ Support/WLAN/StatusBarApp.app/Contents/MacOS/wpa_supplicant`

If you need **wpa_cli** or **wpa_passphrase** you can knock me.

MuntashirAkon, 2019-02-13:
It looks like EAP-TTLS is supported!
[image]

Steve-luo, 2019-02-14:
@MuntashirAkon Thanks so much for the info! I'll do some more digging soon when I have more time.

aleung, 2019-11-12:
@MuntashirAkon 

I face the similar issue of missing EAP method support from the GUI utility. 

I saw that `wpa_supplicant` is running with a temp configuration file and the configure file is regenerated every time when I operate on the GUI utility. Probable I can bypass the GUI utility and run `wpa_supplicant` by myself then use `wpa_cli` to control it. 

But I can't find a `wpa_cli` binary for macOS. Could you give me some hint?

MuntashirAkon, 2019-11-12:
@aleung You can compile it yourself here: https://github.com/MuntashirAkon/hostap_realtek_osx