chris1111/Wireless-USB-Adapter #2: RTL8188

ChadG92, 2018-11-11:
Hello, I was wondering if you've found a way to get the RTL8188FTV working? The device ID is F179 with 0BDA vendor ID. Thank you!

ChadG92, 2018-11-11:
I honestly don't know. I bought it on eBay and it claimed to be Mac OS compatible. The link is here:https://www.ebay.com/itm/NEW-2018-Mini-USB-WiFi-WLAN-Wireless-Network-Adapter-802-11-Dongle-RTL8188-lapto/253793716386?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649

chris1111, 2018-11-11:
> I honestly don't know. I bought it on eBay and it claimed to be Mac OS compatible. The link is here:https://www.ebay.com/itm/NEW-2018-Mini-USB-WiFi-WLAN-Wireless-Network-Adapter-802-11-Dongle-RTL8188-lapto/253793716386?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649

Hum... no CD no Drivers;  strange

closed 2018-11-11 by chris1111