chris1111/Wireless-USB-Adapter #4: StatusBarApp.app is crashing on 10.14.4 beta 2

MuntashirAkon, 2019-02-13:
**StatusBarApp.app** is constantly crashing whenever I click on it on the status bar. I cannot connect to any network using this app.

**System.log**
```
Feb 13 17:03:25 Muntashirs-MBP StatusBarApp[24170]: MenuDelegate::[menuNeedsUpdate]:: numberOfItems=8, VAvailableNetworkItemsDict=1
Feb 13 17:03:25 Muntashirs-MBP StatusBarApp[24170]: MenuDelegate::[menuWillOpen] scheduledScanTimer Thread
Feb 13 17:03:27 Muntashirs-MBP StatusBarApp[24170]: DEPRECATED USE in libdispatch client: dispatch source activated with no event handler set; set a breakpoint on _dispatch_bug_deprecated to debug
Feb 13 17:03:31 Muntashirs-MBP com.apple.xpc.launchd[1] (com.realtek.utility.statusbar.87976[24170]): Service exited due to SIGBUS | sent by exc handler[24170]
```

**Diagnostic Report**
```
Process:               StatusBarApp [24170]
Path:                  /Users/USER/Downloads/*/StatusBarApp.app/Contents/MacOS/StatusBarApp
Identifier:            com.realtek.utility.statusbar
Version:               5.0.5.b2 (1)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           StatusBarApp [24170]
User ID:               501

Date/Time:             2019-02-13 17:03:31.400 +0600
OS Version:            Mac OS X 10.14.4 (18E184e)
Report Version:        12
Anonymous UUID:        D37BDBE7-9EF6-758B-E457-EA153AB25D4F

Time Awake Since Boot: 98000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x000000009adf38b8
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Bus error: 10
Termination Reason:    Namespace SIGNAL, Code 0xa
Terminating Process:   exc handler [24170]

VM Regions Near 0x9adf38b8:
    __TEXT                 000000009abeb000-000000009ac00000 [   84K] r-x/r-x SM=COW  /System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore
--> __TEXT                 000000009ac00000-000000009ae00000 [ 2048K] r-x/rwx SM=COW  /System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore
    Submap                 000000009ae00000-00000000a7e00000 [208.0M] r--/rwx SM=SHM  machine-wide VM submap

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.QuartzCore          	0x9acf79e2 CA::Render::Context::add_begin_time(double) + 94
1   com.apple.QuartzCore          	0x9abee470 CA::Context::commit_transaction(CA::Transaction*) + 3748
2   com.apple.QuartzCore          	0x9abecf2a CA::Transaction::commit() + 528
3   com.apple.AppKit              	0x91618038 __65+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayRefresh]_block_invoke + 452
4   com.apple.CoreFoundation      	0x9389fe1b _runLoopObserverWithBlockContext + 22
5   com.apple.CoreFoundation      	0x93892eea __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 22
6   com.apple.CoreFoundation      	0x93892e13 __CFRunLoopDoObservers + 459
7   com.apple.CoreFoundation      	0x93839e52 __CFRunLoopRun + 2365
8   com.apple.CoreFoundation      	0x93839218 CFRunLoopRunSpecific + 584
9   com.apple.CoreFoundation      	0x93851f7d CFRunLoopRunInMode + 82
10  com.apple.HIToolbox           	0x92e5c8b1 RunCurrentEventLoopInMode + 323
11  com.apple.HIToolbox           	0x92e5c47c ReceiveNextEventCommon + 438
12  com.apple.HIToolbox           	0x92e5c2a7 _BlockUntilNextEventMatchingListInModeWithFilter + 71
13  com.apple.AppKit              	0x915200b0 _DPSNextEvent + 1164
14  com.apple.AppKit              	0x9151ec7b -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 957
15  com.apple.AppKit              	0x9151e8b6 -[NSApplication(NSEvent) nextEventMatchingMask:untilDate:inMode:dequeue:] + 108
16  com.apple.AppKit              	0x91ccc307 _NSHLTBMenuEventProc + 185
17  com.apple.HIToolbox           	0x92edd61f IsUserStillTracking(MenuSelectData*, unsigned char*) + 347
18  com.apple.HIToolbox           	0x930415a4 TrackMenuCommon(MenuSelectData&, unsigned char*, SelectionData*, MenuResult*, MenuResult*) + 1664
19  com.apple.HIToolbox           	0x9303dd33 PopUpMenuSelectCore(MenuData*, Point, double, Point, unsigned short, unsigned int, unsigned int, Rect const*, unsigned short, unsigned long, Rect const*, Rect const*, __CFDictionary const*, __CFString const*, OpaqueMenuRef**, unsigned short*) + 1669
20  com.apple.HIToolbox           	0x9303d189 _HandlePopUpMenuSelection8(OpaqueMenuRef*, OpaqueEventRef*, unsigned long, Point, unsigned short, unsigned int, unsigned int, Rect const*, unsigned short, Rect const*, Rect const*, __CFDictionary const*, __CFString const*, OpaqueMenuRef**, unsigned short*) + 432
21  com.apple.HIToolbox           	0x92ecba9c _HandlePopUpMenuSelectionWithDictionary + 324
22  com.apple.AppKit              	0x918a9bd1 _NSSLMPopUpCarbonMenu3 + 5767
23  com.apple.AppKit              	0x918a84a6 -[NSCarbonMenuImpl popUpMenu:atLocation:width:forView:withSelectedItem:withFont:withFlags:withOptions:] + 508
[... 320 lines cut ...]
IOKit                             7940K        1 
Image IO                           132K        2 
Kernel Alloc Once                    8K        1 
MALLOC                            52.0M       40 
MALLOC guard page                   32K        8 
MALLOC_LARGE (reserved)            256K        1         reserved VM address space (unallocated)
Memory Tag 242                      12K        1 
OpenGL GLSL                        128K        2 
Stack                             10.5M        6 
Stack Guard                       56.0M        6 
VM_ALLOCATE                        128K       17 
__DATA                            12.4M      216 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                        80.1M        7 
__OBJC                            3268K       80 
__TEXT                           157.7M      221 
__UNICODE                          564K        1 
mapped file                      328.8M      205 
shared memory                     2720K       13 
===========                     =======  ======= 
TOTAL                            750.7M      862 
TOTAL, minus reserved VM space   750.4M      862 

```

chris1111, 2019-02-13:
I test lots a App in this Beta and  lots of them crashing so its relate to the Beta I think

Use release Mojave with no issue 
Wait for Apple fix

MuntashirAkon, 2019-02-13:
> I test lots a App in this Beta and lots of them crashing so its relate to the Beta I think

It probably is. However, the error bellow indicates something's changed about the File System. Similar happened during HS-era which rendered many apps unusable. Let's hope it's a bug.

```
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.200.3/Common/ChunkCompression.cpp:49: Error: unsupported compressor 8
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.200.3/Libraries/CompressData/CompressData.c:353: Error: Unknown compression scheme encountered for file '/System/Library/CoreServices/CoreTypes.bundle/Contents/Resources/Exceptions.plist'
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.200.3/Common/ChunkCompression.cpp:49: Error: unsupported compressor 8
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.200.3/Libraries/CompressData/CompressData.c:353: Error: Unknown compression scheme encountered for file '/System/Library/CoreServices/CoreTypes.bundle/Contents/Library/AppExceptions.bundle/Exceptions.plist'
Bus error: 10
```

closed 2019-02-13 by chris1111

MuntashirAkon, 2019-02-20:
Fortunately, working fine on beta 3.

chris1111, 2019-02-24:
Please test this release 👍🏻
Uninstall your package first. Thanks
https://github.com/chris1111/Wireless-USB-Adapter/releases/tag/V1
I test on my MacBook Pro Retina late 2013 (SIP activate) Working perfect 😎