chris1111/Wireless-USB-Adapter #6: Difference between repos

MaximKiselev, 2019-04-15:
hi, thanks for your tools.
What is diff between https://github.com/chris1111/Wireless-USB-Adapter and https://github.com/chris1111/USB-Wireless-Utility.  Im going to use it with tplink usb adapter.

chris1111, 2019-04-15:
> hi, thanks for your tools.
> What is diff between https://github.com/chris1111/Wireless-USB-Adapter and https://github.com/chris1111/USB-Wireless-Utility. Im going to use it with tplink usb adapter.

Realtek drivers and Mediatek drivers

MaximKiselev, 2019-04-15:
oh...i see... its weird. i think you separated tool on two packages kexts + app...thank a lot. One moment - The drivers for tplink based on official drivers or u write them by self? screenshot panel as very similar as official..

chris1111, 2019-04-15:
> oh...i see... its weird. i think you separated tool on two packages kexts + app...thank a lot. One moment - The drivers for tplink based on official drivers or u write them by self? screenshot panel is a very similar as official..

All panel is exactly the same for TP-Link or Asus , D-link, Belkin, Netgear, Ralink, Edimax

closed 2019-04-15 by chris1111