chris1111/Wireless-USB-Adapter #13: Can't use it with multiple USB wifi adapters

stephenfournier, 2019-08-04:
I have multiple USB wifi adapters that I'm trying to use to connect to various wifi networks. The reasoning behind this is I'm using [dispatch-proxy](https://github.com/alexkirsz/dispatch-proxy) to consolidate the networks and increase speed.

I have tried using a combination of these wifi adapters: TL-WN725N, ED-MS1559 and ZTESY AC600, but they all seem to use the same software, REALTEK Wireless Utility (aka "StatusBarApp"). I'm able to connect only 1 of the adapters to a network, and am unable to connect the rest of them. I've also tried running multiple instances of the StatusBarApp, however that had no effect.

chris1111, 2019-08-04:
you can only connect one adapter at a time

closed 2019-08-04 by chris1111

stephenfournier, 2019-08-04:
Is it possible to add this feature from a technical standpoint? Or is it a limitation of Mac?

chris1111, 2019-08-04:
if this is possible in Windows with the same adapters I will not see why it would not be possible but should contact to companies such as TP-Link, Edimax, Trendnet, D-Link to make the Developper add this feature for macOS Application (StatusBarApp)

stephenfournier, 2019-08-04:
I have Windows on bootcamp partition and am able to connect multiple same adapters without issue. I just tested your Wireless USB Adapter-V5 with TL-WN725N and it's working fine. Could you add the feature to this? I can provide a bounty $.

chris1111, 2019-08-04:
I cannot add the feature because I not the developer of the app