chris1111/Wireless-USB-Adapter #14: TL-WN823Nv2 does not work on macOS 10.15 beta5

ctwdtw, 2019-08-07:
@chris1111 : I test the V5 .pkg file on macOS 10.15 beta5 for TL-WN823Nv2, the status bar shows the icon of USB - wifi, but it indicate that no usb wifi device was plug on.

TerryRFred, 2019-08-07:
Ok thanks but where is the file stored on a Mac?

Terry Frederick
Sent from iphone

ctwdtw, 2019-08-07:
@chris1111 , @TerryRFred : It works! I guess the previous installation failure is due to a bug of Catalina beta 5 itself.  During the installation process, I was asked to allow app from identified developers, but when I did it, it seems that Catalina beta5 crashed and backed to the login screen. After uninstall the files, I found a way to successfully install the adapter:
1. unlock the Lock icon to allow identified app developer
2. Then Catalina beta5 will crash and back to login screen
3. login again do step1 again, Catalina will not crash this time.
4. follow the standard process to install the V5 .pkg file, then it works!

Thank you for this wonderful project, my WIFI just broken yesterday and I was on Catalina beta5.
You saved my day!

closed 2019-08-07 by ctwdtw

x43x61x69, 2019-10-14:
Just FYI that TL-WN725N v2 works with v6 on Catalina final version (19A583).

However, I can't get the available networks to show. Manually enter the SSID and password is the only way. (But anything is better as the official RTL driver no longer works with Catalina.)