chris1111/Wireless-USB-Adapter #19: 'This package is incompatible with this version of macOS' (Catalina)

PeteWilliams, 2019-10-09:
Hi,

Your documentation says your drivers work with OSX Catalina, however when trying to install them I get an error saying 'This package is incompatible with this version of macOS'. 

The installer then aborts.

chris1111, 2019-10-09:
> Hi,
> 
> Your documentation says your drivers work with OSX Catalina, however when trying to install them I get an error saying 'This package is incompatible with this version of macOS'.
> 
> The installer then aborts.

Did you see a package that not say This package is incompatible with this version of macOS
on MacOS Catalina

The package works well issue is on your side

closed 2019-10-09 by chris1111

PeteWilliams, 2019-10-09:
> Did you see a package that not say This package is incompatible with this version of macOS
> on MacOS Catalina

I'm not sure what you mean. Could you clarify please?

chris1111, 2019-10-09:
> > Did you see a package that not say This package is incompatible with this version of macOS
> > on MacOS Catalina
> 
> I'm not sure what you mean. Could you clarify please?

I mean that all the packages have the same warning in macOS Catalina because the disk is read-only unless you set it to read

What kind of mac Do you have to use a USB key? it must be an old mac ?

I Will show you a video on real mac book pro 13P Retina late 2013

PeteWilliams, 2019-10-09:
My apologies - I made a mistake. I mixed up two packages that I have downloaded. This one works fine and I have it working now, thanks.

chris1111, 2019-10-09:
> My apologies - I made a mistake. I mixed up two packages that I have downloaded. This one works fine and I have it working now, thanks.

No problem I understand 😄
 I have edit the post with a video on MacOS Catalina from my MacBook Pro