chris1111/Wireless-USB-Adapter #32: RTL8188CU not recognized?

tonyvsuk, 2019-12-11:
Hello,

I've got a USB adapter that identifies in Windows as a RTL8188CU (the readme file lists this as compatible). But it's not recognized by the driver (doesn't detect it). It shows up in IORegistryExplorer though.

Am I doing something wrong? I've got inject kexts set to "yes" in Clover, not sure what else I can do.

Thanks in advance,

Tony.

tonyvsuk, 2019-12-11:
Closed as it worked when I tried a different USB port.

closed 2019-12-11 by tonyvsuk

chris1111, 2019-12-11:
> Closed as it worked when I tried a different USB port.

What is your adapter ?

tonyvsuk, 2019-12-12:
It's just a cheap USB dongle I must have bought from ebay years ago that I had spare in a box, thought I'd try to use it on my hack until the real wifi card turns up.

I've also just realised that I should have posted this in the Wireless-USB-Adapter-Clover section (I'm using that installation and not this one, too many windows open at the same time :) ).

It's working for me, but I do need to remove and reinsert it to get it to work. But it's a cheap USB thing and only temporary so I'm really happy.

Thanks for your work on this.

Tony.

chris1111, 2019-12-12:
Wireless-USB-Adapter as nothing to do with windows open at the same time
So don't open issue for this kind of problem, this is relates to a few things else.

tonyvsuk, 2019-12-12:
I meant I had too many browser windows opened when trying to get the driver working, so entered the issue in the wrong github section. It's nothing to do with the driver or windows, it's me having too many websites open in my browser while trying to get things working.

Thanks again for your work.

Tony.