chris1111/Wireless-USB-Adapter #33: Problems with Archer T3U

VictorCandido, 2019-12-14:
Hi, I'm having some problems with my wifi adapter and its drivers. I tried to install the last version of the drivers just like it is showed in the video, but it didn't worked to me. My Mac still not recognizing the Tp-Link adapter. What should I do?

I've installed the V8 driver and reboot my system. I've already changed the USB port, connected and disconnected many times... But it is still not working.

My adapter: Tp-Link Acher T3U
My SO: Mac OS 10.15.2 Catalina
My Macbook: MacBook Air (Early 2014)

Can anyone help, please? I really need to have my Wifi working as soon as possible.

Thanks.

chris1111, 2019-12-14:
Uninstall and try the V7 Release

VictorCandido, 2019-12-15:
I have installed and uninstalled all the versions, but nothing happened.... Still not working :/

chris1111, 2019-12-16:
Dont know what's wrong with the Tp-Link Acher T3U

closed 2019-12-16 by chris1111