chris1111/Wireless-USB-Adapter #34: Can't connect to 5GHz networks but they appear in the list

error10, 2019-12-21:
I have Catalina 10.15.2 on a Hackintosh ThinkPad T440p and a no-name Chinese 802.11AC adapter based on RTL8812BU chipset, with Vendor ID 0x0bda and Product ID 0xb812.

This adapter appears in Network System Preferences as "Ethernet Adaptor (en2)" instead of a Wi-Fi adapter, and I can only connect to 2.4GHz networks. It can see both 2.4GHz and 5GHz AC networks, but only connected successfully to 2.4GHz networks. If I connect to a 5GHz network then it tries for a few seconds and a red icon appears briefly in the menu bar, as shown in the screenshot.

The adapter works perfectly on 5GHz in Windows and identifies as a Realtek 8812BU 802.11ac USB NIC.

[image]

chris1111, 2019-12-22:
Every WIFI USB adapter is recognize as Ethernet on macOS Catalina 10.15 and I don't know why? No one knows why yet?
For your 5GHz connection I do not have the answer and I do not know if this is possible with such an old WIFI application from D-Link

Raizelian, 2019-12-24:
Hello, I have an adapter with the same chipset you're using, also a chinese brand; it was recognized for me too as "ethernet adaptor (en9)" but I just changed its name to something else. It works perfectly for me on 10.5.2 Catalina, I'm currently connected to a 5GHz network. Might be an issue related to you using Hackintosh, maybe...

closed 2019-12-27 by chris1111