chris1111/Wireless-USB-Adapter #37: RTL8188EUS Support

ann0see, 2019-12-30:
I just connected a LogiLink WL0145A and it seems to get recognised (en3 in network preferences). But there is no connection possible. Are there plans to support this chip?

closed 2019-12-30 by ann0see

chris1111, 2019-12-30:
WL0145A is support by the program

ann0see, 2019-12-30:
Just the Blue LED didn’t blink. That was the problem.

chris1111, 2019-12-30:
Uninstall the programm, reboot and try release V7 Please
thanks let me known

chris1111, 2019-12-31:
Ok I just check the driver for your utility and its not the same
sorry my bad
[image]

ann0see, 2019-12-31:
The latest version does work. I‘m sorry for opening the issue. Probably it was due to an other WiFi dongle which was still plugged in.