chris1111/Wireless-USB-Adapter #43: Mac Won't boot after install

moatazhamada, 2020-02-17:
Mac Catalina 10.15.3 (Hackintosh) after installing the v8 it asked for restart and didn't boot after that and takes forever to load and wont start at the end either

chris1111, 2020-02-17:
Works good here! this is strange ??

closed 2020-02-19 by chris1111

moatazhamada, 2020-02-19:
can you help me with that ? i can access the mac files from dual boot i even deleted the files was installed by the app but it still wont boot

chris1111, 2020-02-19:
The programme is not responsable of an unbootable macOS
Its not on the cause sorry but find your issue, How can I help you ?

moatazhamada, 2020-02-19:
I dont know but it really happened when the installation asked for restart

mlostekk, 2020-04-10:
Same here. I have forgot to disable `SIP`. Now my mac does not boot anymore. Any tips?

mlostekk, 2020-04-10:
I am on mojave by the way

chris1111, 2020-04-10:
> I am on mojave by the way

Boot in recovery HD 
Open terminal type
csrutil disable

Then reboot maybe works

mlostekk, 2020-04-10:
Well, it was disabled when i installed your tool. 

And tried that to boot with disabled mode. Unfortunately it does not work.

mlostekk, 2020-04-10:
Any other idea except recovering from a backup?

chris1111, 2020-04-10:
Reinstall a good macOS because this is already test with a  successful result including me on my MacBook Pro
 I receive a hundred mail by month 
Your Mac run with an issue

pppaolo, 2020-06-27:
I installed it with crsutil enabled apparently. Now it doesnt boot, it get stuck at login bar. Tried any possible safe mood booting.
I also went to recovery mode, terminal and typed csrutil disable. 
Please help.

chris1111, 2020-06-27:
> I installed it with crsutil enabled apparently. Now it doesnt boot, it get stuck at login bar. Tried any possible safe mood booting.
> I also went to recovery mode, terminal and typed csrutil disable.
> Please help.

I have SIP Enable on my MacBook Pro late 2013 and I never have any issue to reboot
No one is responsabble for your MacBook

pppaolo, 2020-06-27:
Thanks for responding so quickly!
I have a macbook pro 2018 I believe.
Never had an issue before.
I installed the wireless usb driver, last version from your github without
disabling SIP.
Restarted the mac and its stuck at 100% loading bar.

What do you suggest I do?