chris1111/Wireless-USB-Adapter #44: some contribution

Airomeo, 2020-02-20:
hi

here is some resources for you.Including some icons and language translation
Realtek WirelessUtility Translation(Simplified Chinese)
dark and light and other icons 

[https://drive.google.com/open?id=1YS4PWLW0R09enLngL4Mxp-Kjcr369VMc](url)

thanks

chris1111, 2020-02-20:
> hi
> 
> here is some resources for you.Including some icons and language translation
> Realtek WirelessUtility Translation(Simplified Chinese)
> dark and light and other icons
> 
> [https://drive.google.com/open?id=1YS4PWLW0R09enLngL4Mxp-Kjcr369VMc](url)
> 
> thanks

Your link not work

chris1111, 2020-02-20:
thanks I will add to the app

closed 2020-02-20 by chris1111

chris1111, 2020-02-20:
edit ********

Release V9 IS remove  because bug report here
https://github.com/chris1111/Wireless-USB-Adapter/issues/45

Airomeo, 2020-02-21:
Sorry for the wrong release.

There is no need to put InfoPlist.strings and MainMenu.nib into the zh-Hans.lproj. Now I have removed.
And here is the updated translation localizable file. It works well in my Mac OS 10.15.3.

[image]

chris1111, 2020-02-22:
I have Add your translation to the front page
[[zh-Hans for Wireless Utility](https://www.notion.so/zh-Hans-for-Wireless-Utility-a31dd055f5be4db4ae6be58f36559bc2)]([zh-Hans for Wireless Utility](https://www.notion.so/zh-Hans-for-Wireless-Utility-a31dd055f5be4db4ae6be58f36559bc2))

Airomeo, 2020-02-22:
Oh bro, you are very welcome. 
You can add it to the next release after your test. Just copy it and paste to StatusBarApp.app/Contents/Resources/zh-Hans.lproj/Localizable.strings. that's all.

Thank you very much.

seekhust, 2020-03-22:
Hi,
I have installed the wireless USB adapter V11 for catalina 10.15.2.

I found it's interesting.

I can see the hardware of the USB wifi under the system report-hardware-USB device tree.

But The USB WIFI adapter can't be recognized on the tool bar. When I click the wifi icon, it say "No USB-WIFI Plug in"

How it happened?

THanks.