chris1111/Wireless-USB-Adapter #45: [Bug] Wireless USB Adapter-V9 && Clover-V12

HsOjo, 2020-02-20:
* When I use this driver V8, it works normally. (on 10.15.3)

However, when updating to V9 or clover V12, the following occurs.
[image]

After that, I try to find the exception; When I compare files, I found the problem.

* /Library/Application Support/WLAN/StatusBarApp.app/Contents/Resources/zh-CN.lproj/InfoPlist.strings

```
/* Localized versions of Info.plist keys */

VersionString= "5.0.7 (2018/02/23)";

DriverName = "All"; /Library/Application Support/WLAN/StatusBarApp.app/Contents/Resources/zh-CN.lproj/Localizable.strings
```

When I delete this file, It works normally again.

HsOjo, 2020-02-20:
There is another problem. When I upgrade my system to 10.15, ISTAT menus cannot monitor the download traffic of the device.

* This is true for all versions of the driver.

[image]

If I use other wired external Ethernet devices, they all work well.

Maybe some interface haven't implements? ? ?

Thanks.          :)

chris1111, 2020-02-20:
I roll back to V8 thanks

closed 2020-02-20 by chris1111