chris1111/Wireless-USB-Adapter #47: WiFI icon

RetroAspie, 2020-03-05:
i know it's just an aesthetic thing, but i'd love it if the icon was the same as the default wifi icon as it fits with the other icons in the menu bar.

if there's a way to change it manually, such as in the .app files then i'll edit this with a method for anyone trying to do the same

chris1111, 2020-03-05:
> i know it's just an aesthetic thing, but i'd love it if the icon was the same as the default wifi icon as it fits with the other icons in the menu bar.
> 
> if there's a way to change it manually, such as in the .app files then i'll edit this with a method for anyone trying to do the same
> 
> edit: i found a method that changes most of the icons
> https://github.com/yunyu/OSXRealtekWifiIcons
> would prefer if this was stock, but this is good enough for anyone who wants to change this at the moment.

Probleme is on Mojave and Catalina the Dark Mode not work on this kind of icons

closed 2020-03-05 by chris1111

lightblair, 2020-03-06:
There's two sets of icons, one for light mode and one for dark. I use dark mode on Catalina and I can see the icons.

chris1111, 2020-03-06:
> There's two sets of icons, one for light mode and one for dark. I use dark mode on Catalina and I can see the icons.

Yes Two set but you have to change Icon for changing Mode Appearance