chris1111/Wireless-USB-Adapter #48: No 5Ghz WiFi speed with Edimax AC1200 (EW-7822ULC)

vblinden, 2020-03-05:
Hello everyone,

First thanks for making the process of installing a WiFi driver for macOS Catalina on the so easy. I bought the Edimax AC1200 (EW-7822ULC) to have Wireless Internet on my laptop. But I can't connect with any 5Ghz network. 2.4Ghz is working fine.

I tried installing the driver version 1.0.2.0 from [here](https://www.edimax.com/edimax/download/download/data/edimax/global/download/for_home/wireless_adapters/wireless_adapters_ac1200_dual-band/ew-7822ulc), but this also doesn't let me connect to 5Ghz networks? I can only connect to 2.4Ghz networks.

What am I doing wrong, am I missing something?

chris1111, 2020-03-06:
Its a known issue and it doesn't seem to work, sorry.

closed 2020-03-06 by chris1111

vblinden, 2020-03-07:
Thanks @chris1111  for clarifying.