chris1111/Wireless-USB-Adapter #50: Generic dongle

Gab627, 2020-03-12:
I have a Mac with Mojave, and a generic usb dongle. I checked the cd drivers that came with it, and it won't do anything. The dongle is recognized in system information, but won't show on network settings.
It seems that is a RTL8188EUS, because of what i saw on the linux drivers from the CD.
Is there any solution?
Thanks for your time

chris1111, 2020-03-12:
Please post the package of CD Drivers here

Gab627, 2020-03-12:
The file was too big to upload here.
https://drive.google.com/open?id=19S71bwVBJPspFfgAKzqsMfEUfIhTPNru

chris1111, 2020-03-13:
Sorry this is not support, this driver stop working at MacOS X Lion 10.7

See driver
[image]

closed 2020-03-13 by chris1111