chris1111/Wireless-USB-Adapter #54: HELP: No USB WiFi Plugin

Vanyoo, 2020-03-25:
Hello bro, I need help...

I use hackintosh, having WN823N, and installed your latest version driver.

[image]
when I plug in the device, OS can detect it. But the driver cannot...

And my SIP status:
`System Integrity Protection status: unknown (Custom Configuration).

Configuration:
        Apple Internal: disabled
        Kext Signing: disabled
        Filesystem Protections: disabled
        Debugging Restrictions: disabled
        DTrace Restrictions: disabled
        NVRAM Protections: disabled
        BaseSystem Verification: disabled
`
(I using opencore as bootloader)

Vanyoo, 2020-03-25:
[image]

chromer030, 2020-03-25:
As it's shown in your screenshot , your adapter is Mediatek Ralink based chipset.
This driver is for Realtek based chipsets.

Vanyoo, 2020-03-25:
> https://github.com/chris1111/Wireless-Ralink-Panel-Utility

Thank you so much, let me try it!

closed 2020-03-25 by Vanyoo