chris1111/Wireless-USB-Adapter #59: Open up my Mac security and install your driver?

gcallah, 2020-04-10:
Why should anyone think this is not a backdoor?

chromer030, 2020-04-10:
All the kexts are signed by Realtek and by modifying/renaming , adding/removing data to/from them , the signature will be lost.
And about utility , its originaly writen by Relatek , and if you dont trust this utility from this package you can use utility released by vendors like Asus , TP-link , Edimax , etc... all are same.

chris1111, 2020-04-10:
> Why should anyone think this is not a backdoor?

Because is Parano

closed 2020-04-10 by chris1111

chris1111, 2020-04-11:
> Why should anyone think this is not a backdoor?
Why did you say this, you did not read the guide
Here is an installation on macOS Catalina 10.15.4 with SIP activated
[image]