chris1111/Wireless-USB-Adapter #60: Very much spam log's. How to disable them?

truepatch, 2020-04-12:
`log show --predicate 'process == "kernel"' --last 3m`

```
kernel: (RtWlanU) chnl_GetBWFrom_IE ChnlBW=0, ExtChnlOffset=0, bEnableVHT = 0, bCurrentHTSupport = 1, bCurrentVHTSupport = 0,
kernel: (RtWlanU) chnl_GetBWFrom_IE ChnlBW=0, ExtChnlOffset=0, bEnableVHT = 0, bCurrentHTSupport = 1, bCurrentVHTSupport = 0,
kernel: (RtWlanU) chnl_GetBWFrom_IE ChnlBW=0, ExtChnlOffset=0, bEnableVHT = 0, bCurrentHTSupport = 1, bCurrentVHTSupport = 0,
```

chris1111, 2020-04-12:
Go to Realtek and for me its not really a spam Log

closed 2020-04-12 by chris1111

chris1111, 2020-06-02:
Fix
https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V15

chris1111, 2020-06-02:
Solved
https://github.com/chris1111/Wireless-USB-Adapter/releases/tag/V12