chris1111/Wireless-USB-Adapter #66: Wifi keeps freezing

mobiyan, 2020-06-18:
When I test the connection with speedtest, I loose wifi during SEND at about a quarter in. I have to disable wifi and re-enable.

Mac Pro (2019)
3.3 Ghz 12-Core
96GB RAM
Radeon Pro W5700X

Wifi device is NetGear Nighthawk AC1900 Wifi USB Adapter

Merci

mobiyan, 2020-06-18:
Seems stable at 2.4Ghz with slower speed

chris1111, 2020-06-18:
Out of curiosity on what macOS are you?

mobiyan, 2020-06-18:
OSX 10.15.5

mobiyan, 2020-06-18:
Can you keep it open to see what is wrong at higher speed on 5Ghz?

chris1111, 2020-06-18:
> Can you keep it open to see what is wrong at higher speed on 5Ghz?

No problem

7Backwards, 2020-06-18:
Have the same problem, and with 5ghz also

7Backwards, 2020-06-18:
It goes hours without anything wrong happening or after 15 min i lose internet but the utility still says im connected, need to replug and plug the wifi dongle again, if i simply turn off wifi and turn on again it doesn´t work.

chris1111, 2020-06-18:
you should know if the 5ghz worked before Catalina to know if it does not come from the last OS's ?

7Backwards, 2020-06-18:
I´m using mojave, was using a older driver, installed now v12 i will report if the issue keeps happening. What i can say is that i had 2.4ghz before and this never happened, now i have 5ghz and 100% isn´t the router´s fault as i tested it before with other equipments.

7Backwards, 2020-06-18:
And yes, it keeps happening.

chris1111, 2020-06-18:
so maybe you have to check with the manufacturer of the USB wifi dongle if  the is a known issue ?

mobiyan, 2020-06-18:
I have a second wifi ac1200, BrosTrend AC1200, same thing happens.

chris1111, 2020-06-18:
I have 3 USB dongle with stable 2.4Ghz

dukechem, 2020-07-11:
Please try disabling IPv6 with steps below and see if that helps. NOTE: keep track of original values of any settings you change so you can undo later if need be.

https://www.privateinternetaccess.com/helpdesk/kb/articles/do-i-need-to-disable-ipv6-traffic-on-my-mac-computer
You can disable IPv6 traffic using the following instructions:

Go to Apple - > System Preferences -> Network
Select the first network connection you see listed on the left-hand side, then click the Advanced button.
Go to the TCP/IP tab at the top
Beside "Configure IPv6", set it to "Link-local Only"
Click Ok to apply the change
Repeat steps 2-5 for any remaining network connections you see in your Network window.
Reboot your computer, so all changes fully take effect.

chris1111, 2020-07-28:
> Please try disabling IPv6 with steps below and see if that helps. NOTE: keep track of original values of any settings you change so you can undo later if need be. [...]

Thanks Closed

lordkev, 2020-08-05:
This did not solve the issue for me. Even with IPv6 disabled on the interface it still drops out after running iperf for <30 seconds on an 802.11ac network.

petkoivanov, 2020-10-10:
This also did not solve the issue for me. 
Still crashes when joining 5g wifi, fine on 2.4

petkoivanov, 2020-10-11:
I tried drivers from the manufacturer - same exact behavior. i also switch to 'link-local only' and still crashes.
may be it is not necessarily this drivers problem but may be something similar to what $lordkev was saying about USB driver ?!
Let me know if i can provide some logs to help investigate this crash. Thanks in advance.

chris1111, 2020-10-11:
This issue is from the Drivers so you must make a problem request to the manufacturer
Issue close

closed 2020-10-11 by chris1111