chris1111/Wireless-USB-Adapter #73: Help with EnGenius EUB9603H (0x9603, 0x1740)

96yrucrem, 2020-07-12:
Hello @chris1111 ,

I am seeking help to get my "EnGenius EUB9603H" USB WiFi adapter that I've been using on a real 15" Late 2013 MacBook Pro to work in Catalina 10.15.5.

After I upgraded from Mojave to Catalina 10.15.5, the icon for the "EnGenius WLAN Client Utility" has become overlaid with a "no entry" icon (circle with a diagonal line).  That's when I realized it's a 32-bit program.

The kext that has worked for the EUB9603H seems to be /Library/Extensions/RTL8192SU7.kext.

I don't remember for sure but I think that kext came from http://www.engenius-uk.co.uk/products-wireless-usb-eub9603h.html.

Executing `file /Library/Extensions/RTL8192SU7.kext/Contents/MacOS/RTL8192SU7` shows it's a `Mach-O 64-bit kext bundle x86_64`.

It seems that the driver is still loading fine in Catalina as I see the EUB9603H's LED blinking.  When I do `kextstat | grep -i rtl`, I see "com.realtek.driver.RTL8192SU" loaded.  And, when I do `kextunload /Library/Extensions/RTL8192SU7.kext`, the LED would turn off.

I tried installing your latest https://github.com/chris1111/Wireless-USB-Adapter/releases/tag/V12 to see if it may support the EUB9603H.  But, it doesn't look like that's the case from looking at the list of "idProduct"+"idVendor" in its Info.plist.

The "idProduct" and "idVendor" for the EUB9603H are "38403" (0x9603) and "5952" (0x1740).

The EUB9603H has been used since Sierra and the manufacturer has discontinued the product.  I couldn't find any new driver package.

I've tried several things, including modifying the `/Library/Application Support/WLAN/StatusBarApp.app/Contents/Info.plist` to add another entry with the EUB9603H "idProduct" and "idVendor".  That produced `The code contains a Team ID, but validating its signature failed.` in /var/log/system.log.  I don't have Xcode or a developer's profile to try to recompile from source.

I would appreciate if you can help me get my EUB9603H to work with Catalina 10.15.5.

Is there an easy way to make the WLAN Utility from your V12 to recognize the EUB9603H?  With the EUB9603H plugged-in, the original/unmodified "StatusBarApp.app" shows "No USB-WiFi Plug in".

Or, better yet, if your V12 can actually support the EUB9603H, can you help to provide with an updated version I can try?

Thank you very much in advance for your help.

chris1111, 2020-07-12:
No way to make that work, its not the same drivers 
sorry

closed 2020-07-12 by chris1111

96yrucrem, 2020-07-12:
Thank you for the quick reply, @chris1111 .

Is there a place somewhere else on the net you can suggest that may know of or has a WLAN Utility that I can try with my existing RTL8192SU7.kext.?

chris1111, 2020-07-12:
I really don't know but if you find a place tell me I am intersteing to look that.

96yrucrem, 2020-07-13:
OK, I'll let you know should I find something that works.

Thank you, @chris1111 .