chris1111/Wireless-USB-Adapter #80: Outdoor compatible adapter

sweetclimusic, 2020-07-31:
I am looking to connect to wifi in my garage. I am looking at the edu ep-8523.  I saw in another closed issue you were able to tell another user the status of the driver if it would work or not "Generic Dongle"

http://www.szedup.com/support/driver-download/ep-8523-driver/

Could you have a look at the driver and verify if it would work?

Thank You

chris1111, 2020-07-31:
Maybe a Router issue ?

sweetclimusic, 2020-07-31:
Sorry, I don't have the edu ep-8523 yet, it has a usb cable that would allow me to plug in to my mac and I could have the dongle outside.  What I wanted to know if it was possible to see if it was compatible from the driver download from their site?

chris1111, 2020-07-31:
For the edu ep-8523 no but on the Download page you will see a version [Realtek MAC 10.15 Drivers](http://www.szedup.com/support/driver-download/)

 but these are practically the same drivers that I use for Realtek and I don't think it will make a difference

closed 2020-07-31 by chris1111