chris1111/Wireless-USB-Adapter #86: [no issue] For users who use Catalina on a non-supported Mac

wearekolossum, 2020-08-12:
Hey there,

i am not sure if this is worth taking into your READE but i had the following setup:

- Installed Catalina on a non-supported Macbook Pro 8.1 via http://dosdude1.com/catalina/
- Chrome was not working as reported by other users
- Installed your Wirelesss-USB-Adapter (by the way: awesome work, thanks a bunch!)
- WiFi was working again
- Chrome wasn't working
- Installed the SIP Disabler patch from DosDude again (as mentioned by other users who had the Chrome problem)
- Restart - the WiFi USB wasn't working anymore (showing "No device attached")
- But Chrome was working
- Installed the Wireless-USB-Adapter again
- Restart - WiFi USB was working again, Chrome wasn't

I circled around that problem as i couldn't get both to work. After a while i tried your other patch https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter which now works like a charm AND Chrome is also running smoothly. Maybe some other people can be helped by this or maybe i missed something in your documentation which already states this.

However, again, THANKS for your great work!

Take care
Florian

jtreg, 2020-09-15:
Hi Chris, I followed the instructions to use my TP-LINK TL-WN722N on Macbook Pro 2017 with OS 10.15.6 Catalina. After reboot the 802.11 did not appear in list of devices in System Preferences/Network. Not sure where I went wrong here... I installed the V12 package. Any help gratefully requested.

chris1111, 2020-09-15:
> Hi Chris, I followed the instructions to use my TP-LINK TL-WN722N on Macbook Pro 2017 with OS 10.15.6 Catalina. After reboot the 802.11 did not appear in list of devices in System Preferences/Network. Not sure where I went wrong here... I installed the V12 package. Any help gratefully requested.

 
I don't know what you are missing but it is suppose to work with WN722N? SIP is activate or not ?

closed 2020-09-16 by chris1111