chris1111/Wireless-USB-Adapter #88: Can't connect to 5Ghz wifi

CheckHelzio, 2020-08-28:
OS: Catalina 10.15.2
UBS Adapter: Archer T9UH v.2.8
When I try to connect to a 5Ghz wifi the wifi utility crashes and does not recognize the adapter. I need to unplug it and plug it in again for it to work again. It only happens with 5ghz networks.

chris1111, 2020-08-28:
https://github.com/chris1111/Wireless-USB-Adapter/issues/66

closed 2020-08-28 by chris1111

CheckHelzio, 2020-08-28:
Tank you for the fast answer... the solution of disabling IPv6 works for me.

chris1111, 2020-08-28:
> Tank you for the fast answer... the solution of disabling IPv6 works for me.

Thanks to confirm

medoh300, 2021-03-13:
for me the the ipv6 disabling doesnt work , i see 5ghz channels but not all of them , especially the higher ones in the 100s .