chris1111/Wireless-USB-Adapter #89: The icon is too small

z1un, 2020-09-10:
The icon is too small. I hope it can be changed a little bigger, which is consistent with the style of other icons in the system.
[image]

chris1111, 2020-09-10:
> The icon is too small. I hope it can be changed a little bigger, which is consistent with the style of other icons in the system.
> [image]

Make a better one then share 😁

closed 2020-09-10 by chris1111

z1un, 2020-09-10:
Can you make a better icon update？ hahah

chris1111, 2020-09-10:
> Can you make a better icon update？ hahah

I have already try then almost impossible, if you see the Icon its work for Light and Dark Mode and its the only way what i found to make it work

chris1111, 2020-09-10:
try this icon, change then reboot 
[Icon.zip](https://github.com/chris1111/Wireless-USB-Adapter/files/5202865/Icon.zip)

this icon looks more bigger on Dark mode

z1un, 2020-09-10:
thinks

z1un, 2020-09-10:
> 试试这个图标，先更改然后重新启动
> [Icon.zip](https://github.com/chris1111/Wireless-USB-Adapter/files/5202865/Icon.zip)
> 
> 在黑暗模式下，此图标看起来更大

How should I use it?

chris1111, 2020-09-10:
> > 试试这个图标，先更改然后重新启动
> > [Icon.zip](https://github.com/chris1111/Wireless-USB-Adapter/files/5202865/Icon.zip)
> > 在黑暗模式下，此图标看起来更大
> 
> How should I use it?

On the Root Disk ➤/Library/Application Support/WLAN/StatusBarApp.app/Contents/Resources
Copy paste Icon
this is will change the Icon then reboot

z1un, 2020-09-10:
> > > 试试这个图标，先更改然后重新启动
> > > [Icon.zip](https://github.com/chris1111/Wireless-USB-Adapter/files/5202865/Icon.zip)
> > > 在黑暗模式下，此图标看起来更大
> > 
> > 
> > How should I use it?
> 
> On the Root Disk ➤/Library/Application Support/WLAN/StatusBarApp.app/Contents/Resources
> Copy paste Icon
> this is will change the Icon then reboot

okay thinks！