chris1111/Wireless-USB-Adapter #90: TL-WN722N On OS 10.15.6 Macbook Pro 2017

jtreg, 2020-09-15:
Hi Chris, I followed the instructions to use my TP-LINK TL-WN722N on Macbook Pro 2017 with OS 10.15.6 Catalina. After reboot the 802.11 did not appear in list of devices in System Preferences/Network. Not sure where I went wrong here... I installed the V12 package. Any help gratefully requested.

adminy, 2020-09-16:
Same here, stupid update ... I have a windows partition also and there I get a "could not get device descriptors (code 43)".
Where before the update everything worked great.

adminy, 2020-09-16:
Managed to roll back, everything works now. Also I have 3 dead ports, they don't seem to come back even with a nvram reset.

chris1111, 2020-09-16:
> TL-WN722N

I don't know what's wrong here TL-WN722N  suppose to works before and after update 10.15.6 this is already test successfully.

closed 2020-10-09 by chris1111