chris1111/Wireless-USB-Adapter #91: USB stuck on DISK mode

dgonzalezs, 2020-09-21:
Hey, first of all, thanks for all your work.

I bought recently this generic dongle ([https://www.amazon.com/Bluetooth-Adapter-Wireless-External-Receiver/dp/B07YDFZWT8/](https://www.amazon.com/Bluetooth-Adapter-Wireless-External-Receiver/dp/B07YDFZWT8/) and I can't get it to work, it stays as a DISK.

Product ID 0x1a2b
Vendor ID 0x0bda

I'm on a Mid 2011 iMac 21.5 on High Sierra 10.13.6. I tried the drivers on amazon to no luck either ([https://bit.ly/2YA01yM](https://bit.ly/2YA01yM)

Any help will be greatly appreciated.

chris1111, 2020-09-21:
don't use to in keyboard Hub plus to another USB port

dgonzalezs, 2020-09-21:
> don't use to in keyboard Hub plug to another USB port

Yeah, I tried every USB port, it doesn’t matter which I use, it never changes from Disk to NIC

Saludos, 
Diego González S.

dgonzalezs, 2020-09-21:
I managed to get it to work 2 times, but after a while it disconnects and can't reconnect again. But I'm guessing that could be my OS, since the Airport card does the same (that's the reason I bought a USB wireless adapter). Would a fresh installation help with that?

chris1111, 2020-09-21:
Probably ?

closed 2020-10-09 by chris1111