chris1111/Wireless-USB-Adapter #102: Tried everything to make it work but cannot get past "No USB-WiFi Plug-in"

akmant, 2020-12-19:
Hi,

No matter what I do, I cannot make my it work. What is it that I am doing wrong? I tried with v13, v12 and Big Sur versions with no success. I have not managed to get past this void where it says "No USB-WiFi Plug-in" and I am not able to do anything at all. The only thing that has never happened (if it helps, and I don't know how to trigger it) as you have explained in <https://github.com/chris1111/Wireless-USB-Adapter/issues/9>, I have never seen the message in Security stating: "System Software from developer "Realtek Semiconductor Corp" was blocked from loading. And when I look at kexts, they are installed.

sudo kextload /Library/Extensions/RtWlanU.kext -v
Requesting load of /Library/Extensions/RtWlanU.kext.
/Library/Extensions/RtWlanU.kext loaded successfully (or already loaded).

sudo kextload /Library/Extensions/RtWlanU1827.kext -v
Requesting load of /Library/Extensions/RtWlanU1827.kext.
/Library/Extensions/RtWlanU1827.kext loaded successfully (or already loaded).

sudo kextload /Library/Extensions/RtWlanDisk.kext -v
Requesting load of /Library/Extensions/RtWlanDisk.kext.
/Library/Extensions/RtWlanDisk.kext loaded successfully (or already loaded).

You can see all from the pict attached. What is it I am missing?

Thanks for your help and contribution.

[image]

chris1111, 2020-12-19:
you missing to say what ia your adapter brand and model

akmant, 2020-12-19:
It was at the bottom left of the picture but OK. It is and OEM product using Realtek 0xf179 --> RTL8188FTV and no it is not a virtual machine, it is a MacBook Pro 15 Mid 2012; no hacks, no tricks. :)

chris1111, 2020-12-19:
> It was at the bottom left of the picture but OK. It is and OEM product using Realtek 0xf179 --> RTL8188FTV and no it is not a virtual machine, it is a MacBook Pro 15 Mid 2012; no hacks, no tricks. :)
RTL8188FTV is not a brand wifi adapter this drivers is not support by the program

otonieru, 2021-08-20:
Hi @chris1111 is it possible for you to reopen this issue, there's more and more adapter are implementing this RTL8188FU/FTV (Product ID : 0xf179) which currently has no driver for mac. The Linux driver is available though. Can you port it to mac ?

chris1111, 2021-08-20:
> Hi @chris1111 is it possible for you to reopen this issue, there's more and more adapter are implementing this RTL8188FU/FTV (Product ID : 0xf179) which currently has no driver for mac. The Linux driver is available though. Can you port it to mac ?

I am not a dev, I am a Packager but it would be nice if someone could do it

closed 2021-09-03 by chris1111

zakariamehbi, 2021-12-06:
Having the same problem.

A program tried to load new system extension(s) signed by “Realtek Semiconductor Corp” that need to be updated by the developer.

Jrap-bit, 2022-07-28:
Facing the same problem with TP-Link AC600 High Gain Wireless Dual Band USB Adapter Archer T2U Plus. Never encountered the 'System Software from developer "Realtek Semiconductor Corp" was blocked from loading' message. The adapter also not showing up. I have an M1 Pro Macbook 14" 2021.

minksW, 2023-08-19:
i fixed it i guess by reinstalling, noticed that the kext wasnt even in library/extensions
thanks for this cmd: sudo kextload /Library/Extensions/RtWlanU.kext -v