chris1111/Wireless-USB-Adapter #105: A few unlocalized strings in French

cortig, 2021-01-08:
Hi,

I noticed that some of the strings weren’t localized in the French Localizable.strings.
I took the liberty of localizing them (and also took the opportunity to correct a few typographical inconsistencies etc).
The updated file is here:
[French Localization.zip](https://github.com/chris1111/Wireless-USB-Adapter/files/5788034/French.Localization.zip)

As a side note, I peeked into the Canadian-French file and the format looked weird to me… I wonder whether there might be something wrong with it.

Corentin

chris1111, 2021-01-08:
The fr-CA.lproj is the same format as French.lproj 
post here [pull requests](https://github.com/chris1111/Wireless-USB-Adapter/pulls) the changes you want to make

closed 2021-01-08 by chris1111

cortig, 2021-01-21:
I tried cloning the repository to create a pull request, but I’m not familiar enough with GitHub and so far miserably failed to see the code anywhere in the cloned repository in GitHub to add the updated localization. Sorry about that. I’ll try again though…

cortig, 2021-01-21:
In the process, I added the missing strings and corrected a few typographical issues in the CA version as well. I still have no clue how to insert them in the code and create a pull request though :-\

[French and French-Ca localizations.zip](https://github.com/chris1111/Wireless-USB-Adapter/files/5850032/French.and.French-Ca.localizations.zip)

chris1111, 2021-01-21:
> In the process, I added the missing strings and corrected a few typographical issues in the CA version as well. I still have no clue how to insert them in the code and create a pull request though [...]

Pull request then add file ZIP  because people see the file

cortig, 2021-01-21:
Hi Chris,

I’m desperately trying. I suspect the confusion on my end comes from the fact that there’s no code in the repository and I therefore can’t figure out where to add my files. When I try to create a pull request, I get a generic “nothing to compare and can’t get any further. I’m still toying with it, but so far not being very successful.

cortig, 2021-01-21:
I think I might have gotten it… I just created a pull request, adding the two localization .lproj.