chris1111/Wireless-USB-Adapter #125: File missing

Greatisgreat, 2022-01-27:
The only pkg file i can find is WIFI Icon Collor.pkg and I cant find the Wireless USB Adapter.pkg
Is the WIFI Icon Collor.pkg supposed to install the wireless adapter? If so I am getting "The Installation Failed"

Sorry if i am being stupid

chris1111, 2022-01-27:
Link is here [Wireless USB Adapter.pkg](https://github.com/chris1111/Wireless-USB-Adapter/releases/tag/V14)

closed 2022-01-27 by chris1111

Greatisgreat, 2022-01-27:
So is WIFI Icon Collor.pkg the pkg used to install the utility?

chris1111, 2022-01-27:
> So is WIFI Icon Collor.pkg the pkg used to install the utility?

> Link is here [Wireless USB Adapter.pkg](https://github.com/chris1111/Wireless-USB-Adapter/releases/tag/V14)

Link is here Wireless USB Adapter.pkg