chris1111/Wireless-USB-Adapter #128: restarting the computer does not see the WN722NV3 adapter

Lyk5t, 2022-03-04:
There is a problem as in the title, after restarting the computer, WN722NV3 does not see it, insert the adapter, it warns

chris1111, 2022-03-05:
Are you sure you have the V3 ?
V3 is support without issue

closed 2022-03-05 by chris1111

Lyk5t, 2022-03-05:
yes i am absolutely sure i am facing such a problem what can be done about this problem?

chris1111, 2022-03-05:
witch macOS do you have ?

I have this adapter on Monterey 12 without probleme

Lyk5t, 2022-03-06:
the operating system is mac os catalina and I have it as an ethernet adapter in the network section.
It doesn't look like yours. I installed the driver directly, I didn't make any other settings, do I need to make any other settings?

Note: system hackintosh
I am sorry, my English is not good