chris1111/Wireless-USB-Adapter #134: 8188cu doesn't work on macos12.5

zx0922, 2022-08-10:
No usb-wifi plug in

chris1111, 2022-08-10:
what is the brand of the adapter?

zx0922, 2022-08-12:
MERCURY MW150UH

zx0922, 2022-08-12:
> what is the brand of the adapter?

MERCURY MW150UH

chris1111, 2022-08-12:
MERCURY MW150UH is only work on Windows see Support downloads
https://www.mercusys.com/en/download/mw300uh#Software

closed 2022-08-12 by chris1111

zx0922, 2022-08-12:
> MERCURY MW150UH is only work on Windows see Support downloads https://www.mercusys.com/en/download/mw300uh#Software

Thanks for replay.
The link is mw300uh, mine is MW150UH, the chip is rtl 8188gu.

chris1111, 2022-08-12:
No the link is MERCURY MW150UH

see https://www.mercusys.com/en/product/details/mw300uh/

Click support then Software

zx0922, 2022-08-12:
> No the link is MERCURY MW150UH
> 
> see https://www.mercusys.com/en/product/details/mw300uh/
> 
> Click support then Software

[image]

chris1111, 2022-08-12:
Its for Windows

zx0922, 2022-08-12:
> Its for Windows

yes. I mean it's MW300UH. MW150UH 's chip is rtl8188eu.
MW150UH is like this: [image]

zx0922, 2022-08-12:
it's works on macos 10.15. Is it possible work on 12.5? Thanks a lot!

chris1111, 2022-08-12:
macOS 12 and 13 its here
https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter
Try maybe?

zx0922, 2022-08-13:
> 

I've try. No usb-wifi plug in

zx0922, 2022-08-13:
> macOS 12 and 13 its here https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter Try maybe?

I've try. No usb-wifi plug in