chris1111/Wireless-USB-Adapter #140: How to uninstall?

blksith0, 2023-03-22:
We're lacking uninstallation instructions.

chris1111, 2023-03-22:
> We're lacking uninstallation instructions.

Same place as install

closed 2023-03-22 by chris1111

blksith0, 2023-03-23:
Ctrl+F, search "uninstall" --> 0 results.
*gasp*

blksith0, 2023-03-23:
"Each .pkg file generally differ in where they install the files, so unlike a .app bundle, there is no generalized way to uninstall the contents of a .pkg file.

A .pkg file is an installer package. It is run with the help of the macOS built-in [Installer](https://en.wikipedia.org/wiki/Installer_(macOS)) app, and allows the installation of the files contained under the package under various directories as programmed by the developer of the software."

But here's something that might help to uninstall chris1111's drivers: https://www.corecode.io/uninstallpkg/